Chadderz121/bakingpi-www #3: "OK" LED is 17th GPIO pin

muellermartin, 2015-04-21:
I think there is an off by one error in the naming of the OK LED: Since lesson OK01 the "OK" LED is decribed to be the 16th pin, but now I'm pretty sure it actually is the 17th pin, because the GPIO pin numbering starts at zero and of course by looking at all the offsets for setting the GPIO function and the output. Of course the pin is referred to as "[GPIO16][1]" but there is also a "GPIO0" pin.

[1]: http://elinux.org/RPi_BCM2835_GPIOs

Chadderz121, 2015-04-22:
Hmm, this is an interesting one. The debate here is, is the GPIO0 pin the "first (1st)" pin or the "0th" pin. Looking at my descriptions, I also talk about the 16th bit in a register, which implies I have a 0th bit. Perhaps it's worth a clarifying comment to be explicit about what I mean. Alternatively it could be rewritten to say "pin 16" and "bit position 16" with a note somewhere saying that "pin 0 is the first pin". It may even be a nice opportunity to describe off by one errors :wink:

muellermartin, 2015-04-22:
Yes, definitively! I have a programming background, so I'm somewhat used to count from zero, but I guess it can be difficult for newcommers.
I like the idea to use different naming "pin 16" vs. "bit position 16" (although this still is ambigous if there was no prior notice whether there is also a 0th bit/pin). Maybe this can be underlined with an indicator (color, symbol, etc.) to always make clear which counting method is used and a box explaining the meaning of the indicators.
It could be quite useful to explain the off-by-one error, e.g. in the troubleshooting section.