Chadderz121/bakingpi-www #7: Lesson 6 frameBuffer.s Alignment Inconsistency

mfritze, 2015-06-10:
In [Lesson 6](http://www.cl.cam.ac.uk/projects/raspberrypi/tutorials/os/screen01.html) part 4 (My Dearest Graphics Processor), it mentions the importance of using `.align 4` such that the lower 4 bits are 0s. The solution uses `.align 12` as page alignment, justified for correct communication with the GPU. Is this a mistake?

Chadderz121, 2015-06-16:
`.align 4` is correct and sufficient. `.align 12` came from some of my early experiments when I didn't understand the GPU caching behaviour. The solution should be corrected, not the tutorial text. I'll leave this issue open as a reminder to do that.

chaibpt, 2018-04-16:
The bug is still open which is causing the confusion in understanding alignment the concept.