Chadderz121/bakingpi-www #12: OK02 working on Raspberry Pi 2 B (with changes needed)

ExileInParadise, 2015-07-09:
Thanks to the mighty hackers at the Pi forums, I've managed to cobble together enough information to get OK02 working.
https://www.raspberrypi.org/forums/viewtopic.php?f=72&t=114796&sid=1706008439d80e4fe83e7d02401acead

First, the base physical address for SoC peripherals has moved from 0x20000000 to 0x3F000000 on the Pi 2 B.

Second, the GPIO pin for the OK/ACT LED is now 47 rather than the previous 16.
In the post you will see a table showing the peripheral base and gpio pin for the recent models.

Third, the logic of 0 to enable and 1 to disable the LED seems to have been inverted somewhen recently, probably those with 40 pin GPIO header rather than 26 pin.
I am not sure I got that logic correct in my quick and dirty test... it may be backwards in the code below, but the LED still flashes *grin*

Fourth, there is now a kernel.img AND a kernel7.img in the boot area of the SD card for Pi 2 B.
I was able to get the OK02 working with just kernel.img in the boot area for it.

```
/******************************************************************************
*	main.s
*	 by Alex Chadwick
*
*	A sample assembly code implementation of the ok02 operating system, that 
*	simply turns the OK LED on and off repeatedly.
*	Changes since OK01 are marked with NEW.
******************************************************************************/

/*
* .section is a directive to our assembler telling it to place this code first.
* .globl is a directive to our assembler, that tells it to export this symbol
* to the elf file. Convention dictates that the symbol _start is used for the 
* entry point, so this all has the net effect of setting the entry point here.
* Ultimately, this is useless as the elf itself is not used in the final 
* result, and so the entry point really doesn't matter, but it aids clarity,
* allows simulators to run the elf, and also stops us getting a linker warning
* about having no entry point. 
*/
.section .init
.globl _start
_start:

/* 
* This command loads the physical address of the GPIO region into r0.
*/
ldr r0,=0x3F200000 /* NEW */

/*
* Our register use is as follows:
* r0=0x3F200000	the address of the GPIO region.
* r1=1 << 21 	a number with bits 21-23 set to 001 to put into the GPFSEL4 NEW
*				function select to enable output to GPIO 47. NEW 
* then
* r1=1 << 15	a number with bit 15 high, so we can communicate with GPIO 47. NEW
* r2=0x003F0000 a number that will take a noticeable duration for the processor 
*				to decrement to 0, allowing us to create a delay.
*/
mov r1,#1
lsl r1,#21 /* NEW GPFSEL4 bit 21 to enable GPIO47 for out */

/*
* Set the GPIO function select.
*/
str r1,[r0,#0x10] /* NEW GPFSEL4 instead */

/* 
* Set the 15th bit of r1.
*/
mov r1,#1
lsl r1,#15 /* NEW we want 15th bit for GPSET1 and GPCLR0 */

/* NEW
* Label the next line loop$ for the infinite looping
*/
loop$: 

/*
* Set GPIO 47 to low, causing the LED to turn on.
*/
str r1,[r0,#0x20] /* NEW GPSET1 instead of GPSET0 */

/* NEW
* Now, to create a delay, we busy the processor on a pointless quest to 
* decrement the number 0x3F0000 to 0!
*/
mov r2,#0x3F0000
wait1$:
	sub r2,#1
	cmp r2,#0
	bne wait1$

/* NEW
* Set GPIO 47 to high, causing the LED to turn off.
*/
str r1,[r0,#0x2C] /* NEW GPCLR1 instead of GPCLR0 */

/* NEW
* Wait once more.
*/
mov r2,#0x3F0000
wait2$:
	sub r2,#1
	cmp r2,#0
	bne wait2$

/*
* Loop over this process forevermore
*/
b loop$
```

and the kernel.list from the assembly:
```

build/output.elf:     file format elf32-littlearm

Disassembly of section .init:

00008000 <_start>:
    8000:	e59f003c 	ldr	r0, [pc, #60]	; 8044 <wait2$+0x10>
    8004:	e3a01001 	mov	r1, #1
    8008:	e1a01a81 	lsl	r1, r1, #21
    800c:	e5801010 	str	r1, [r0, #16]
    8010:	e3a01001 	mov	r1, #1
    8014:	e1a01781 	lsl	r1, r1, #15

00008018 <loop$>:
    8018:	e5801020 	str	r1, [r0, #32]
    801c:	e3a0283f 	mov	r2, #4128768	; 0x3f0000

00008020 <wait1$>:
    8020:	e2422001 	sub	r2, r2, #1
    8024:	e3520000 	cmp	r2, #0
    8028:	1afffffc 	bne	8020 <wait1$>
    802c:	e580102c 	str	r1, [r0, #44]	; 0x2c
    8030:	e3a0283f 	mov	r2, #4128768	; 0x3f0000

00008034 <wait2$>:
    8034:	e2422001 	sub	r2, r2, #1
    8038:	e3520000 	cmp	r2, #0
    803c:	1afffffc 	bne	8034 <wait2$>
    8040:	eafffff4 	b	8018 <loop$>
    8044:	3f200000 	.word	0x3f200000
```

ExileInParadise, 2015-07-12:
The same changes work for OK03.

Effectively there are at least 3 programming configurations:
Raspberry Pi 1 A or B - peripheral base 0x20000000, use GPIO pin 16 for OK/ACT
Raspberry Pi 1 A+ or B+ - peripheral base 0x20000000, use GPIO pin 47 for OK/ACT
Raspberry Pi 2 B - peripheral base 0x3F000000, use GPIO pin 47 for OK/ACT

I am assuming the Pi 1 A+ works the same as the B+, due to having the same 40 pin GPIO as the Pi 1 B+ and Pi 2 B.
I have no idea what changes might be needed for the compute modules.

Also, I don't know whether to update these pages to show 3 different configurations in the appropriate places... or whether the entire tutorial should be split into 3 paths:
Baking Pi (for the Pi 1 A and B)
Baking Pi+ (for the Pi 1 A+ and B+)
Baking Pi 2 (for the Pi 2 B)

ExileInParadise, 2015-07-12:
The same changes work for OK04 as well on Pi 2 B.
The systemTimer.s works with the peripheral base 0x3F000000 change.
Add the same timer base 0x00003000 offset as a Pi 1.

ExileInParadise, 2015-07-12:
And just to finish out - OK05 works as well on Pi 2 B using peripheral base 0x3F000000 and GPIO47.
Using the bit flashing pattern it does seem like the use of 1 and the 0 have been inverted on the Pi 2 B (and possibly the Pi1A+ and Pi1B+).
During the section of the bit string where there were a number of zeroes, the LED seemed to stay lit for long periods.

jonndavis1993, 2016-02-23:
I am try to run the code above on my Raspberry Pi 2 B and I cannot get the ACT LED to blink. I have follow the tutorial to the letter and that code also does not work. I have tried modifying the code to try and get it to work but I still am unable to get the LED to blink. Is there something you did to get the code to work?

What happens when I load the kernal.img file to the RPi, the ACT LED lights up and stays solid green. Maybe I am not generating the img file correctly. The current process I am using is (on window's 10) going to the source folder and then opening up cmd.exe and running the 'make' command. It generate the kernal.img file. I use the Win32DiskImager to write the file onto the SD. Any tips you can offer, or any links that will point me in the write direct would be greatly appreciated.

viveksingh123321, 2016-04-28:
please mention the build command for it in a little bit detailed manner

jonndavis1993, 2016-05-05:
The build command I use is the same command listed in the baking pi tutorial. I open the command prompt window under the same directory I have the files and I enter 'make'. No errors are returned.

viveksingh123321, 2016-05-06:
Thanks for the reply

 I got it working. But in the similar way I tried to make GPIO 0(GPIO 17 )
by just modifying the address of the register. Is it possible in this way
or do I need some more modifications ?

jonndavis1993, 2016-05-06:
I do not understand what you mean by "But in the similar way I tried to make GPIO 0(GPIO 17 )
by just modifying the address of the register". Could you please clarify?

viveksingh123321, 2016-05-09:
I am saying that by changing the address, your code also works fine for
raspberry Pi2.

jonndavis1993, 2016-05-11:
@ viveksingh123321 Is your code exactly the same as the code posted by @ExileInParadise? If so, could you tell me the set up of your development machine. I want to know where my config differs. If you code is different than the code above could you tell me exactly where you made changes?

viveksingh123321, 2016-05-12:
I have attached the code file. Please take a look

jonndavis1993, 2016-05-18:
@viveksingh123321 I do not see the code file. Are you referring to the link to the github page for Baking-Pi?

arthurptj, 2016-07-05:
does anyone have the Mailbox addresses for Pi 2 B that coincide with the screen chapters of the tutorial?
Thank you

Bottersnike, 2016-12-02:
Did you ever get round to updating the site? I've always liked it as a nice simple way to start to learn assembly.