Chadderz121/bakingpi-www #16: lesson 8: screen 03 on Raspberry Pi 2

7asper, 2015-10-22:
Hello,

I tried to set up screen03 on RPI2, but it is not working.
Framebuffer might be initialized correctly, because the monitor is set to black. But no letter appears.

Do you have an Idea, @edargelies

edargelies, 2015-10-22:
Hi,
Yes, I had the same issue. It seems on the RPi2 adding the font.bin data causes the frame buffer data to be moved to an inaccessible location. I explain how I got it working on the last post on this thread https://www.raspberrypi.org/forums/viewtopic.php?f=72&t=17502&sid=d113045f7e40130136f286f0d494ca59 . Let me know if you still have issues.
Eric

7asper, 2015-10-22:
I'm not sure anymore:

1. What kind of Linux do you use as a base?
2. What kind of config.txt do you use?

edargelies, 2015-10-23:
I've been in contact with 7asper, but for anyone else following.
1. I'm using the official Raspbian image as a base
2. The configuration file is default, all properties are commented out.