Chadderz121/bakingpi-www #29: Add information for 3B+ Model

rishacha, 2018-06-16:
I was able to able to get the `OK01`module up and running for Pi 3B+ by going through the manual. 

I got to know through an online [post](https://www.raspberrypi.org/forums/viewtopic.php?t=208381) about the GPIO Pin for Activity LED being 29.

Can I contribute and add the necessary snippets?

ExileInParadise, 2019-01-12:
Please do - either in comments or through a github pull request.

floese, 2020-11-02:
> I was able to able to get the `OK01`module up and running for Pi 3B+ by going through the manual.
> 
> I got to know through an online [post](https://www.raspberrypi.org/forums/viewtopic.php?t=208381) about the GPIO Pin for Activity LED being 29.

I'd love to see your code adapted for Pi 3B+. Can I see it somewhere?