Chadderz121/bakingpi-www #32: Raspberry Pi3b+

rdr-1999, 2019-01-10:
Please include information on how to proceed building the OS for higher versions like Raspberry Pi 3b+. The OK01 lesson isn't working as the ACT light is not glowing on my pi

ExileInParadise, 2019-01-12:
Changes will need to be made to adapt these Pi1 A/B tutorials for Pi2 and Pi3.
The first change is that the memory mapped I/O space moved from 0x2000000 on early Pi up to 0x3f000000 to allow for the larger memory being added to these.
Second, the GPIO header went from 26 to 40 pins, which resulted in the OK/ACT LED moving from GPIO 16 to GPIO47 (i think).
I think there may be some more code changes will be needed when it comes to the screen series tutorials.
Please see Issue 12 and Issue 16 for more tips on how to get these working.
Also take a look at the excellent bare metal code examples from Dave Welch also here on github and referred to be the bare metal forum on the Raspberry Pi site.
It is cool to see more people interested in these tutorials, and with some experimentation and research, as well as a lot of testing, hopefully we can figure out how to update these for newer Pi and add more and more OS features....

NeoFahrenheit, 2019-03-22:
I would like it too. :)

I'm doing Computer Science in my university here in Brazil and I would love to see an update in this tutorial. I think my biggest difficulty would be to make the Pi "see and execute" my code.

I guess that would be setting the Program counter to the code, the stack and data pointer correctly...
I'm tryng to the see the processor manual to see where all these adresses are. :)

Thank you. :D

mdvthu, 2022-01-07:
Has anyone managed to identify the precise memory addresses for the Raspberry Pi 4 LED (not just an estimate: I think 0x2000000 is possibly out by a factor of 10??)?

In the tutorial, there is this additional unexplained caveat that means I really don't know where to start when trying to figure out the memory mapped addresses on the Pi 4: _For added confusion, the manual uses a different addressing system. An address listed as 0x7E200000 would be 0x20200000 in our OS._